JuliaPhysics/BeamletOptics.jl #11: Jesus bug: Small angle spot diagram abnormalities

StackEnjoyer, 2025-04-17:
> [!WARNING]  
> To reproduce this MWE, the branch [#10](https://github.com/StackEnjoyer/BeamletOptics.jl/pull/10) is required, or later (merged) versions due to the `PointSource` type.

# System setup

For the following MWE some spurious effects can be observed when tracing a point source with very small spread angle onto a `Spotdetector` with a tilted first lens:

```julia
using GLMakie, BeamletOptics, LinearAlgebra

const BMO = BeamletOptics

hide_axis(ls::LScene) = (ls.show_axis[] = false)

numerical_aperture(θ::Real, n::Real=1.0) = n*sin(θ)
airy_disk(λ::Real, NA::Real) = 0.61*λ/NA

##
l1 = SphericalLens(48.88e-3, 182.96e-3, 8.89e-3, 52.3e-3, λ -> 1.62286)
l23 = SphericalDoubletLens(36.92e-3, Inf, 23.06e-3, 15.11e-3, 2.31e-3, 45.11e-3, λ -> 1.58565, λ -> 1.67764)
l45 = SphericalDoubletLens(-23.91e-3, Inf, -36.92e-3, 1.92e-3, 7.77e-3, 40.01e-3, λ -> 1.57046, λ -> 1.64128)
l6 = SphericalLens(1063.24e-3, -48.88e-3, 6.73e-3, 45.11e-3, λ -> 1.62286)
l_23 = BeamletOptics.thickness(l1) + 0.38e-3
l_45 = l_23 + BeamletOptics.thickness(l23) + 9.14e-3 + 13.36e-3
l_6 = l_45 + BeamletOptics.thickness(l45) + 0.38e-3
translate3d!(l23, [0, l_23, 0])
translate3d!(l45, [0, l_45, 0])
translate3d!(l6, [0, l_6, 0])

# dealign the first lens by a small amount
xrotate3d!(l1, deg2rad(0.5))

sz = 5e-3
pd = Spotdetector(sz)
translate_to3d!(pd, [0, 0.147, 0])

system = StaticSystem([l1, l23, l45, l6, pd])
```

When solving the system with the following `PointSource`, the resulting spot diagram looks normal:

```julia
fig = Figure()
ax = Axis(fig[1,1], aspect=1, xlabel="x [mm]", ylabel="y [mm]")

# normal ring trace
ps = PointSource([0, -0.5, 0], [0,1,0], deg2rad(2), 486.0e-9, num_rays=5000, num_rings=50)

empty!(pd)
solve_system!(system, ps)

scatter!(ax, pd.data, markersize=3, color=:red)

sleep(5) # necessary for next plot  to render correctly ? idk why
```

Note how a single spot lies outside of the grouped spot diagram. This is the center beam...

[image]

# The bug

When tracing the same system again with a very narrow `PointSource`, the abnormal spot diagram can be generated:

```julia
ps = PointSource([0, -0.5, 0], [0,1,0], 5e-5, 486.0e-9, num_rays=5000, num_rings=3)

empty!(pd)
solve_system!(system, ps)

spots_weird = copy(pd.data)

scatter!(ax, spots_weird, markersize=3, color=:blue)
```

[image]

This spot pattern does not appear correct at all. 

# Potential bug sources

Of the potential sources, the `refraction3d` method has been considered first. For very small angles, the function might produce faulty direction vectors. The test case strictness has been overhauled with [this commit](https://github.com/StackEnjoyer/BeamletOptics.jl/pull/10/commits/cd7c6a0c48119492ce6e0244db8f9ec3b87ace72). The test passes for the current implementation of `refraction3d`. It is therefore unlikely, that this is the error source. Further sources might be:

- [ ] `sdf` of the spherical lenses
- [ ] errors introduced during rotation of the lens element via `_world_to_sdf`

TacHawkes, 2025-04-18:
I think I have a clue on this one. I suspect the [numeric_gradient](https://github.com/StackEnjoyer/BeamletOptics.jl/blob/31d6560148084c3f510ad4a1dd028a9bc73a20e2/src/SDFs/AbstractSDF.jl#L103) to be the culprit. Try changing the eps to a smaller (or larger) value to see some effect on the 'weird' spot pattern. For 1e-10 I get a somewhat circular pattern. It also removes the singular outlier.

My questions:
- Is this solvable by picking a different default value, does this lead to bugs/implications elsewhere 
- Can we improve how we generically calculate normals? See, e.g. https://iquilezles.org/articles/normalsSDF/

TacHawkes, 2025-04-18:
A possible fix is now part of https://github.com/StackEnjoyer/BeamletOptics.jl/pull/10

closed 2025-04-18 by StackEnjoyer